penfold42/ESPixelBoard #12: ESPixelStick, ESP8266 (NodeMCU) and Hyperion (Kodi)?

CaptInsano, 2017-12-06:
I am looking to have my Kodi/SPMC setup on my Nvidia Shield control LEDs connected to an ESP8266. I currently use a Raspberry Pi (running Hyperion) & arduino nano elsewhere in the house. 

After hours of searching for a clear answer on the hyperion forums I ended up flashing the ESPixelStick firmware to my NodeMCU and I have connected 300 WS2812b LEDS to it. I can access the webserver on the NodeMCUand run the "Testing" colours through my LED strip but how do I get Kodi to control the LEDs? I have read that this firmware allows UDP control but I have not seen any instructions how to get the control up and running.

Any help would be greatly appreciated!

penfold42, 2017-12-06:
The upstream repository only supports the udp e.131 sacn protocol but my fork also adds a raw udp protocol on port 2801.

If you are using classic Hyperion, your only choice is the udpraw protocol

If you are using Hyperion.ng you can use either udpraw or e.131

I recommend Hyperion.ng because it’s got a much nicer web UI and if you use udpe131 you pretty much just need to configure the up address of your nodemcu

If you’re still stuck please open a topic on https://hyperion-project.org/forums/software-support.10/ so more people can find the discussion

CaptInsano, 2017-12-11:
Ahh, I understand now.

Excuse me and thanks for your help!

closed 2017-12-11 by CaptInsano