penfold42/ESPixelBoard #17: Q: Will this run on a H801?

Moonbase59, 2018-07-20:
I got a few cheap ESP8266-based H801 RGB2W light controllers (https://www.aliexpress.com/item/rgbww-strip-WiFi-controller-1-port-control-200-lights-communicate-with-Android-phone-via-WLAN-to/32301423622.html) and was wondering if I could use these with _both_ MQTT and DMX (E1.31).

For some reason, PWM support seems to have been in forkineye’s ESPixelStick but is now gone.

Questions:

1. Would your code run on a H801 device?
2. Would you find the heart to merge forkineye’s latest master branch into yours (after the weekend, because I made a PR with some changes)?

penfold42, 2018-07-20:
1. Yep ! The webUI even includes the h801 labels next to the gpios

2. Yep also - I’m planning to track as closely as I can. The effect updates didn’t merge cleanly so I have some to resolve some conflicts there before I update the mqtt changes from the last few days

Moonbase59, 2018-07-20:
Hey, good news! So I reckon I give it a try as soon as you’ve merged in the newest stuff and I have some time to play with the H801s. :-)

penfold42, 2018-07-20:
Be careful - don’t power it externally AND connect the FTDI serial at the same time.

Moonbase59, 2018-07-20:
Hee hee, of course not ;-) I’ve been "ESPURNA’ing" _lots_ of 230V E27 LED bulbs and power switches lately. Thanks anyway :-)

Moonbase59, 2018-08-12:
Well, I had some spare time to try it on a H801 today:
* Folder needs to be renamed to "ESPixelStick", conflicts with the real one (I renamed `ESPixelStick.ino` to `ESPixelBoard.ino`)
* I have
  ```
  /* Enable support for rotary encoder */
  //#define ESPS_ENABLE_BUTTONS
  ```
  but still get
  ```
  sketch/buttons.cpp:4:27: fatal error: RotaryEncoder.h: No such file or directory
   #include <RotaryEncoder.h>
  ```

penfold42, 2018-10-14:
It _should_ be all good to go

It took a little while to add support for multiframe websocket frames - saving device settings was broken cos of that.

penfold42, 2018-11-02:
Done

closed 2018-11-02 by penfold42

Moonbase59, 2018-11-07:
I’ll give it a try one of these days. Good to see you’re still on it :-)