penfold42/ESPixelBoard #50: Error Compiling in "Serial Mode"

Andreadj, 2019-03-12:
Hi,
sorry I'm new to Arduino and programming. Compiliong the sketch in "Pixel mode" everything goes fine but when compiling in "Serial mode" I've got this:

Arduino:1.8.8 (Windows Store 1.8.19.0) (Windows 10), Scheda:"LOLIN(WEMOS) D1 R2 & mini, 160 MHz, Flash, Disabled, 4M (3M SPIFFS), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

In file included from C:\Users\Ufficio\Desktop\Nuova cartella\Nuovo_da_zero\ESPixelBoard-master\ESPixelStick\ESPixelStick.ino:37:0:

C:\Users\Ufficio\AppData\Local\Temp\arduino_build_689225\sketch\wshandler.h: In function 'void procG(uint8_t*, AsyncWebSocketClient*)':

wshandler.h:278:27: error: 'GAMMA_TABLE' was not declared in this scope

                 gamma.add(GAMMA_TABLE[i] >> 8);

                           ^

Uso la libreria ESPAsyncE131-master alla versione 1.0.1 nella cartella: E:\Privato\Documenti\Arduino\libraries\ESPAsyncE131-master 
Uso la libreria ESPAsyncUDP-master alla versione 1.0.0 nella cartella: E:\Privato\Documenti\Arduino\libraries\ESPAsyncUDP-master 
Uso la libreria ESP8266WiFi alla versione 1.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266WiFi 
Uso la libreria Hash alla versione 1.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\Hash 
Uso la libreria SPI alla versione 1.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\SPI 
Uso la libreria Ticker alla versione 1.0 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\Ticker 
Uso la libreria async-mqtt-client-master alla versione 0.8.2 nella cartella: E:\Privato\Documenti\Arduino\libraries\async-mqtt-client-master 
Uso la libreria ESPAsyncTCP-master alla versione 1.2.0 nella cartella: E:\Privato\Documenti\Arduino\libraries\ESPAsyncTCP-master 
Uso la libreria ESP8266mDNS alla versione 1.2 nella cartella: E:\Privato\Documenti\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0\libraries\ESP8266mDNS 
Uso la libreria ESPAsyncWebServer-master alla versione 1.2.0 nella cartella: E:\Privato\Documenti\Arduino\libraries\ESPAsyncWebServer-master 
Uso la libreria ArduinoJson alla versione 5.13.5 nella cartella: E:\Privato\Documenti\Arduino\libraries\ArduinoJson 
Uso la libreria RotaryEncoder-master alla versione 1.1.0 nella cartella: E:\Privato\Documenti\Arduino\libraries\RotaryEncoder-master 
exit status 1
'GAMMA_TABLE' was not declared in this scope

Any idea?

penfold42, 2019-03-12:
Yep, it’s my fault !

It might be a few days before I get to fixing it

Thanks for reporting it

Andreadj, 2019-03-13:
Ok, thanks!